turbobabr/easyeda-ibom-extension #2: Wrong position on some highlighted parts

vlrmprjct, 2021-03-15:
Looks like some parts have a mirrored position.
I dont know is it an EasyEDA Issue or a extension issue.

See Screens:
[image]
[image]

xsrf, 2021-03-15:
I noticed the same, usually on connectors. Looks like the highlight frame is always centered around the component origin, so it is offset for components which don't have their origin in their center.

[image]
[image]

turbobabr, 2021-03-16:
Wow! "Dark" guys are here! :) Thank you for reporting the bug!

@xsrf Yup.. that's the line, just fixed it. The updated version is here:
- https://github.com/turbobabr/easyeda-ibom-extension/releases/download/v0.0.7/easyeda-ibom-v0.0.7.zip

closed 2021-03-16 by turbobabr