turbobabr/easyeda-ibom-extension #12: footprint completely missing - pads misshaped

xsrf, 2021-06-06:
Hey,
the attached footprint/pcb won't work at all. The footprint is just a oval shaped pad and no highlight. This might be partially due to #9 as it has no silkscreen but also the pads are wrong.
I'll try to debug this more.

[image]

[image]

[PCB_ibom missing footprints2_2021-06-06.zip](https://github.com/turbobabr/easyeda-ibom-extension/files/6604074/PCB_ibom.missing.footprints2_2021-06-06.zip)

turbobabr, 2021-06-06:
Some pad shapes are not supported yet. I suspected that this footprint contains pads based on polygons. Will investigate the issue in a couple of days. Thank you for providing all the required data.

xsrf, 2021-06-06:
@turbobabr yeah, it uses `POLYGON` pads of which one is zero width. That's why one is oval (default) and one is missing. The actual visible pads are `SOLIDREGION` which also is not supported - even outside of footprints.

So far I tried adding the mapping to `custom` for the pad shape and supplying a `svgpath` attribute but it's not working.
However, I got `SOLIDREGION` outside of footprints to work but I'm not sure where to integrate it. I mocked it inside the top/bottom zones.
My understanding of what is done where isn't that huge at the moment ;)
I'll tinker around a bit more

turbobabr, 2021-06-06:
No worries about this one - I will add support for all the pad shapes during the week.

xsrf, 2021-06-06:
Great 🥳
But for this also the non `PAD` things are missing. There are lots of footprints that use tracks or solid regions etc. I think, each `FOOTPRINTS`s `TRACKS`, `SOLIDREGION`, `SVGNODE` etc. should be just merged into the PCBs equivalents for drawing the PCB preview. Guess this is pretty easy if done at the right spot ;) 

For the bounding box, `TRACKS`, `SOLIDREGION`, `SVGNODE` etc. of each footprint must also be considered. But for the highligted Pads just using `PAD` should be fine. It will still fail on this one, as it has a pad with zero width, but that's the footprints problem.

turbobabr, 2021-06-06:
> But for this also the non PAD things are missing. There are lots of footprints that use tracks or solid regions etc. I think, each FOOTPRINTSs TRACKS, SOLIDREGION, SVGNODE etc. should be just merged [...]

This is an issue of trying to adapt `InteractiveHTMLBom` to work with EasyEDA. KiCad has a different data model and it's a bit restricted comparing to EasyEDA. My initial intention was to come up with a quick solution to hand-assemble a bunch of board -> had no idea anyone else going to use this extension.

I'm going to come up with some sort of "canonical" PCB document that covers all the cases. Any help with covering the variety of possible cases will be greatly appreciated! :)

xsrf, 2021-06-06:
I don't think it's that complex ;) In EasyEDAs datamodel each footprint basically has the same properties as a PCB itself. A footprint has `SOLIDREGION`, `TRACK` etc. just as the PCB. For drawing the preview you don't actually care if it is a footprint, but you wanna see it, so merging them should be fine, I guess. 

Only the Pads are special...

[image]

turbobabr, 2021-06-08:
@xsrf, The majority of mentioned issues, included the one with the  footprint example you provided are fixed in this version:
- [easyeda-ibom-v0.1.0.zip](https://github.com/turbobabr/easyeda-ibom-extension/releases/download/v0.1.0/easyeda-ibom-v0.1.0.zip)

Please let me know whether it work for you.

xsrf, 2021-06-08:
Thx, for the footprint in question it works fine 👍

I'll open another issue for the missing solid regions and tracks within footprints

closed 2021-06-08 by xsrf