turbobabr/easyeda-ibom-extension #14: Solid region shapes aren't rendered at all

turbobabr, 2021-06-08:
[image]

[iBom_solid_regions_rendering_issues_2021-06-08.json.zip](https://github.com/turbobabr/easyeda-ibom-extension/files/6616357/iBom_solid_regions_rendering_issues_2021-06-08.json.zip)

closed 2021-06-08 by turbobabr

xsrf, 2021-06-08:
You were faster than me opening the issue ;) However, I think you missed filtering the `SOLIDREGION` for `type: "solid"`. There is also `type: "cutout"` which shouldn't be rendered as copper area.

turbobabr, 2021-06-08:
Ugh... completely forgot about additional modes - thank you for pointing out. Added a filter in this release:
- https://github.com/turbobabr/easyeda-ibom-extension/releases/download/v0.1.2/easyeda-ibom-v0.1.2.zip