turbobabr/easyeda-ibom-extension #15: Copper features inside footprints not rendered

xsrf, 2021-06-08:
Footprints can have their own Tracks, solid regions or Images (SVGNodes) potentially containing copper or silkscreen. At the moment these are not rendered at all which makes some footprints hard to identify on the pcb preview.
I think these features should just be rendered as if they were part of the PCB itself.

Since the datamodel of these features is the same as for the features on the PCB itself, they could just be copied from the footprint to the PCB data before rendering.

[image]

turbobabr, 2021-06-08:
Managed to do some work with additional shape types, guessing there will be bugs... :)

Available here:
- https://github.com/turbobabr/easyeda-ibom-extension/releases/download/v0.1.2/easyeda-ibom-v0.1.2.zip

xsrf, 2021-06-08:
That actually works pretty good already 👍

I've noticed `CIRCLE` support is also missing completely 🙈

turbobabr, 2021-06-08:
> That actually works pretty good already 👍
Awesome! 🚀 

> I've noticed CIRCLE support is also missing completely 🙈
Yeah.. `CIRCLE`s and `ARC`'s are missing atm -will add them a bit later.

closed 2021-06-08 by turbobabr

xsrf, 2021-06-08:
Made a horrible example 😄

[image]
[image]

turbobabr, 2021-06-08:
Could you share the JSON source for this one? Will add it to my "canonical" tests.

xsrf, 2021-06-08:
Sure... the tilted part is actually the footprint `__TEST_IBOM_ISSUE`

[PCB_ibom_2021-06-08.json.txt](https://github.com/turbobabr/easyeda-ibom-extension/files/6617971/PCB_ibom_2021-06-08.json.txt)

turbobabr, 2021-06-08:
Cool! Thanks!