turbobabr/easyeda-ibom-extension #17: update local.txt

EasyEDA2021, 2021-06-11:
Can you update the local.txt? I would like to help tp update the i18n

thanks

turbobabr, 2021-06-11:
It might be a little be tricky since actual `iBom` representation is ran in an independent `iFrame` and I am not sure whether EasyEDA does wildcard replacement of the text occurrences listed in the `locale.txt`. But yeah, I could try to add additional `on the fly` processing for those files.

Also, there might be an issue with how `Chinese Simplified` localization is going to look, since the original `InteractiveHTMLBom` implementation is targeted for `en` locale only.

Anyway, feel free to send `locale.txt` with all the strings listed in there.

turbobabr, 2021-06-11:
**Unrelated:**

Are you the one who is responsible for maintaining these lists of extensions?
- https://easyeda.com/forum/topic/Extension-User-Extensions-for-EasyEDA-Summary-9e065b68316f4491a3911dc6204be31e
- http://club.szlcsc.com/article/details_49807_1.html

EasyEDA2021, 2021-06-11:
> 
> 
> **Unrelated:**
> 
> Are you the one who is responsible for maintaining these lists of extensions?
> 
>     * https://easyeda.com/forum/topic/Extension-User-Extensions-for-EasyEDA-Summary-9e065b68316f4491a3911dc6204be31e
> 
>     * http://club.szlcsc.com/article/details_49807_1.html

Yes, I am :)

turbobabr, 2021-06-11:
Cool! I have a suggestion on how to `"semi-automate" the maintenance of that list. A while ago I've build this registry like repo:
- https://github.com/turbobabr/easyeda-extension-registry

The core of this thing is `extensions.json` file which is sort of `"database"` for known extensions:
- https://github.com/turbobabr/easyeda-extension-registry/blob/main/extensions.json

Here are some benefits of using such a repo instead of manually built list on the forum:
- All the extensions are listed as a single `json` file, thus it can be processed or interpreted in many ways.
- Each listed extensions might has all the required data including locales and all.
- Users can submit their extensions without reaching the people at EasyEDA using pull requests.
- Pull request notifications might be used a `"signal"` to update content on forums or other places. The repo even contains [special scripts](https://github.com/turbobabr/easyeda-extension-registry/blob/b5fc978b7c871efb93c021d1c1d10e0b6ea8ea23/package.json#L8) for quick building of lists in markdown format.

What I'm talking about is such registry is a `single source of truth` for all the known extensions with an ability to add new extensions for other developers. In case it sounds interesting, I can add you as a collaborator to the repo or even transfer the ownership of the repo to an official `EasyEDA` github account.

In a mean time, I suggest replacing those cluttered tables you used both in EasyEDA and LCEDA forums with compact lists like these:

**EasyEDA (EN):**

- [EasyEDA-Tools](https://github.com/ppeetteerrs/EasyEDA-Tools) by Peter Yuen: A set of commands like reusable modules(module renaming + replacing net names), global font replace and net names replacing
- [iBom for EasyEDA](https://github.com/turbobabr/easyeda-ibom-extension) by Andrey Shakhmin: Use InteractiveHTMLBOM right within EasyEDA!
- [Label Maker](https://github.com/xsrf/easyeda-labelmaker) by Andreas Bernhofer: Create beautiful labels with solid background for your PCBs.
- [Main Menu UX Normalizer](https://github.com/turbobabr/easyeda-extension-menu-ux-normalizer) by Andrey Shakhmin: Removing annoying menu popovers on mouseover event.
- [PCB Tools](https://github.com/dmwatson/easyeda-pcbtools) by dmwatson: Track Width calculator, which can quickly calculate your needed track width based on amperage and layer type
- [QR Code Generator](https://github.com/turbobabr/easyeda-qrcode-generator-extension) by Andrey Shakhmin: Generate and add QR Codes to PCB.
- [Quick Align](https://github.com/turbobabr/easyeda-quick-align-extension) by Andrey Shakhmin: Objects alignment without a hassle!
- [Smooth Tracks](https://github.com/mattshepcar/easyeda-round-tracks) by Matt Shepcar: Rounds your track corners and adds nicer teardrops
- [SVG Import](https://github.com/xsrf/easyeda-svg-import) by Andreas Bernhofer: Import SVG files to PCB without any distortions!
- [Themes](https://github.com/FiercestT/EasyEdaThemes) by Fiercest: Additional Light and Dark themes for EasyEDA.
- [Toggle View](https://github.com/xsrf/easyeda-toggleview) by Andreas Bernhofer: Switch PCB view between Top and Bottom view.

**LCEDA (CN):**

- [EasyEDA-Tools](https://github.com/ppeetteerrs/EasyEDA-Tools) by Peter Yuen: 修改原理图字体，修改原理图模块网络标签，修改网络名
- [iBom for EasyEDA](https://github.com/turbobabr/easyeda-ibom-extension) by Andrey Shakhmin: 编辑器本地版InteractiveHtmlBom
- [Label Maker](https://github.com/xsrf/easyeda-labelmaker) by Andreas Bernhofer: 快速创建反相文本丝印
- [Main Menu UX Normalizer](https://github.com/turbobabr/easyeda-extension-menu-ux-normalizer) by Andrey Shakhmin: 阻止鼠标移动到顶部菜单就自动下拉
- [PCB Tools](https://github.com/dmwatson/easyeda-pcbtools) by dmwatson: 根据阻抗计算导线宽度
- [QR Code Generator](https://github.com/turbobabr/easyeda-qrcode-generator-extension) by Andrey Shakhmin: 输入文本创建高清二维码图片
- [Quick Align](https://github.com/turbobabr/easyeda-quick-align-extension) by Andrey Shakhmin: 元素快速对齐
- [Smooth Tracks](https://github.com/mattshepcar/easyeda-round-tracks) by Matt Shepcar: 创建圆弧型泪滴；布线转为圆弧布线
- [SVG Import](https://github.com/xsrf/easyeda-svg-import) by Andreas Bernhofer: 插入高清SVG图片到PCB
- [Themes](https://github.com/FiercestT/EasyEdaThemes) by Fiercest: 修改整个编辑器界面颜色
- [Toggle View](https://github.com/xsrf/easyeda-toggleview) by Andreas Bernhofer: 翻转板子底层视图，镜像PCB

Both of these lists were generated automatically from the `extensions.json` file available in [easyeda-extension-registry](https://github.com/turbobabr/easyeda-extension-registry/blob/main/package.json#L8) repo, using [these scripts](https://github.com/turbobabr/easyeda-extension-registry/blob/b5fc978b7c871efb93c021d1c1d10e0b6ea8ea23/package.json#L8).

EasyEDA2021, 2021-06-11:
Hi
Thanks for the suggestion
we have been think about this, but we don't have more resource to do this, we are developing a new pro edition(still have free plan), and it will be released maybe in three months later, base on the webGL, and we will design a new extension system at pro edition in the future
as before, our API and extension feature are less users to use, then we didn't make it better, at these months we found these great extension, then we want to share them to other users ASAP, post at the forum is quick.
Thank you for your great extension.