turbobabr/easyeda-ibom-extension #24: iBOM doesn't save HTML files

jonchandler, 2022-12-30:
Awesome extension which is very useful!

I want to make an HTML reference library of iBOMs for my boards for easy reference, but when I click "Generate and save HTLM", no file is saved, even without an extension.

I'm using EasyEDA V6.5.9 on Windows 11 Home.

Any suggestions?  Thanks,

jonchandler, 2023-02-01:
I have discovered something about this issue.  The board where generating and downloading an HTML file fails is a board created in Eagle and imported into EasyEDA..  It does not have an associated schematic, but it does have a BOM.  Displaying the iBOM page works beautifully, but downloading does not work.

With EasyEDA created schematic&boards, the generate and download function works beautifully.  Awesome addition to EasyEDA.

zl4430, 2023-02-27:
me to
I'm using EasyEDA V6.5.9 on Windows 11 Home.
期待升级版